cnlohr/ws2812esp8266 #8: Add to Lua project ?

bigmonkeyboy, 2015-01-21:
Not an Issue per se... but any thoughts on possibility as adding the ws2812 driver to the nodemcu-firmware project ? https://github.com/nodemcu/nodemcu-firmware

It would be a great addition to that if it could be done. thanks

cnlohr, 2015-01-26:
Sadly, I've never had any interest in Lua, and everything I'm doing is with C... I don't really see any use in Lua personally... Though I wouldn't mind if people figure out better ways of doing the wait times and committing it to their project.

closed 2015-01-26 by cnlohr

bigmonkeyboy, 2015-01-26:
Well the contribution would be a driver in C :-)
But not a problem. thanks.

cnlohr, 2015-01-28:
I'm not sure what you mean by that?  It just sounds like a feature request that should be filed to them?

bigmonkeyboy, 2015-01-28:
Yes - exactly... I will. Thanks for the project.