cnlohr/ws2812esp8266 #9: more then 25 leds?

sfranzyshen, 2015-02-04:
I have not been able to get this to work with more then 25 leds ...

cnlohr, 2015-02-12:
I keep putting this off saying "I'll get to testing it tomorrow" but it just keeps not happening!  I want to ask if you've checked a few things first:

(1) Where is your power source?  If USB, that will only power about 15-20 WS2812's safely.

(2) What is the behavior?  Do the others flash wildly?

(3) Are you sure the power going into your ESP8266 is good?  I.e. are you using a dedicated 3.3V regulator, or are you trying to bank on the regulator in the USB->Serial adapter being good enough?

sfranzyshen, 2015-02-14:
nope. it wasn't any of these things ... it was your timimg code
https://github.com/sfranzyshen/ws2812esp8266

scargill, 2015-03-14:
If you follow sfranzyshen's code and use the 100pf cap... I have 300 LEDs running. I modified the code to allow for variable output pins and a variable buffer length with repetition...

scargill, 2015-03-15:
On the other hand I am getting LED1 flickering occasionally and right now I can't figure out why - but I do have 2 outputs running 2 LED displays in the background under a 50ms callback, running the colour wheel from the original Arduino library and all is well.. just need to figure out why that first LED flickers.

sfranzyshen, 2015-03-16:
Hi scargill! I also had problems with the first led flicker ... I solved it with a level shift on the data line ...

scargill, 2015-03-16:
Yes, it’s REALLY annoying but it can be solved with timing  - it’s that reset pulse – I’m doing the update now – see http://tech.scargill.net in about half an hour. Ta.

 

Pete.

cnlohr, 2015-10-08:
This is now a known issue with interrupts in the newer SDKs.  This message is here as a not to anyone finding it: Please migrate to esp8266ws2812i2s

closed 2015-10-08 by cnlohr