cnlohr/ws2812esp8266 #12: Use of PWM pins

billprozac, 2015-03-05:
The GPIO register commands you use don't seem to work to well on the PWM pins (GPIOs 12, 13, 15).
If you replace:
WRITE_PERI_REG( PERIPHS_GPIO_BASEADDR + GPIO_ID_PIN(WSGPIO), 1 );
with
gpio_output_set(GPIO_BIT_ID, 0, GPIO_BIT_ID, 0);

where GPIO_BIT_ID = BIT13 for pin13

Now these pins seem to work fine.

Great lib by the way.

cnlohr, 2015-03-06:
Thanks.  Do you mind just making this change yourself and pushing it to my repo?

j0hncc, 2015-05-21:
It doesn't seem to work for WSGPIO=4 or WSGPIO=5 either ( for GPIO4 and GPIO5)?

cnlohr, 2015-10-08:
Ah! I believe that is because of the multiple functions on the pins.  You have to configure pins to be GPIO in some cases.

This is the code used in esp8266ws2812i2s

		static const uint32_t AFMapper[16] = {
			0, PERIPHS_IO_MUX_U0TXD_U, 0, PERIPHS_IO_MUX_U0RXD_U,
			0, 0, 1, 1,
			1, 1, 1, 1,
			PERIPHS_IO_MUX_MTDI_U, PERIPHS_IO_MUX_MTCK_U, PERIPHS_IO_MUX_MTMS_U, PERIPHS_IO_MUX_MTDO_U };

		int nr = my_atoi( &pusrdata[2] );

		if( AFMapper[nr] == 1 )
		{
			buffend += ets_sprintf( buffend, "!G%c%d\n", pusrdata[1], nr );
			return buffend - buffer;
		}
		else if( AFMapper[nr] )
		{
			PIN_FUNC_SELECT( AFMapper[nr], 3);  //Select AF pin to be GPIO.
		}

closed 2015-10-08 by cnlohr