cnlohr/ws2812esp8266 #14: Include files not found at compile time

source83, 2015-09-16:
After running through the steps to set the build environment up and making some slight modifications to adapt to changes, the Makefile can't find the SDK's include files.
The paths in the Makefile are updated and correct, but when compiling, it can't find the header files from the sdk

cnlohr, 2015-09-17:
The specific folder may vary depending on which SDK you use, you will still have to modify the Makefile to have the SDK path be correct.

i.e. Change the following to where it is for your user.
SDK:=/home/cnlohr/esp8266/esp_iot_sdk_v1.2.0

source83, 2015-09-17:
version 1.2.0 ?
Thought we were supposed to use 0.93

source83, 2015-09-17:
found the problem, needed https://github.com/esp8266/esp8266-wiki/blob/master/include.tgz
I've added it to both 0.9.3 and 1.3.0
Problem solved :D

closed 2015-09-17 by source83