cnlohr/ws2812esp8266 #16: Confusing

jonshouse1, 2016-04-26:
Thanks for the fantastic project, I now have this working :-)

I found the following a little confusing ?
"You send a UDP packet to it on port 7777 and it plops it out GPIO0"

According to the pinouts I can find the RX pin is not GPIO0.   Google "esp8266 pinout" to see what I mean.

Also on my hardware the 100pf cap (or any other value upto 10nf) made zero difference. I did need a level shifter to make it work - running the ws2812 chain directly from the GPIO pin was all over the place, but either reducing the ws2812 chain voltage to 3.2v (bench PSU) or sticking a non inverting buffer in cured it. 

What are the maximum number of pixels I can update in one chunk of UDP data?

I have a linux C test program that cycles R/G/B a bit if anyone wants it. 

Thanks,
Jon

cnlohr, 2016-04-26:
I think you're thinking about https://github.com/cnlohr/esp8266ws2812i2s ... this is an old project and less stable than that one (See the notes)

I am surprised you had a problem needing a level shifter, though.  I've never needed one. Also, not sure why, but, with the other project, it seems much more reliable as well as long as the ESP gets a SOLID 3.3V.  Biggest problem I had here was I was trying to run it off a 3.3v supply that couldn't handle it.

I think I set everything up to work upto the network MTU, so ~480 LEDs I think? I never tried.

Charles

jonshouse1, 2016-04-26:
Yes, sorry I had the old and the newer project web pages open as the build environment notes are on the older one.  The code I have built is the newer one.

The LEDs I am using are these:
 http://www.espruino.com/refimages/WS2811_25.jpg

I am using an ESP8266 original board on a USB dev board just for power.  Basically 3.3v from a linear regulator taking 5v from USB and the LEDs from the bench PSU.  If I turn the bench PSU up to 5v (It powers the LEDs only) they become very messy, but if I run the LEDs from 3.3v then all is well.    I cant figure out what is going on, I may need to put a scope on it.  The two PSUs are grounded together ok.  Maybe the first LED board has a pull up to 5v or something - I will check. Todays job was to see if I could make the code work on the module, this is the first time I have built anything for the ESP8266 from source, up until now I have only used the AT interpreter.
I just found the 'top' demo code - I missed that first time round, very pretty :-) 
I ask the limits on LEDs because I am currently using raspberry Pi boards to power 5 house window sized LED matrix built from sets of these LEDs. I will be putting animations on them for my xmas light display. I am surprised how well the wifi module is doing, much smoother than I expected so maybe I will use those instead :-)

Thanks for publishing a set of great projects.
Jon

cnlohr, 2016-04-27:
Those are ws2811's... but, I will try with ws2812's this evening, hopefully and let you know how it goes.

cnlohr, 2016-04-28:
AAHA! Are you using your ESP8266 with the programmer still attached??? If so, make sure your output pin is free to go full swing.  Programmers sometimes drive the pin high or low.  Also, I am closing this issue. All future issues should go to esp8266ws2812i2sr

closed 2016-04-28 by cnlohr

jonshouse1, 2016-04-28:
The programmer I am using lets me pluck RX directly from the module with no pullups or similar.  I am now using the module on a bit of vero strip with a buck converter to provide 3.3v from 5v.  It still requires a level converter to work with the WS2811 chain I have, clearly not a problem with your code ! :-)  Thanks,Jon

jonshouse1, 2016-04-28:
PS The board I was using was well thought out but lacked any documentation. For anyone interested I have documented the links here:
http://www.jonshouse.co.uk/esp8266_ESP-01_ESP-202_devboard-vendor_yison.png

cnlohr, 2016-04-30:
When you say pluck RX from module, do you mean it /also/ disconnects it from the programmer?  If not, it won't work, or will barely work.  BUT! Please clarify with this:  Are you actually using /this/ project or are you using the esp8266ws2812i2s project?  I'm sorry I'm still confused about this... RX is for the i2s project, GPIO0 is for this one I believe.  Also, with this one it can suffer from crashing using larger blocks of data.

jonshouse1, 2016-04-30:
Would you like me to re-open this on the correct project or I can email you directly with some pictures if you prefer?

>When you say pluck RX from module, do you mean it /also/ disconnects it from the programmer?
Yes, exactly. The TX/RX pins are available on the dev board, the links route them from the module to the serial IC, so removing the link completely and adding a connection wire connects directly to the module RX line.

I have now build a unit with a board running on its own, IE an ESP-01 module powered from a 3v buck converter, with a level shifter and the WS2811 LED chain - no development board.

>Please clarify with this: Are you actually using /this/ project or are you using the esp8266ws2812i2s project?
I am using the code here:
https://github.com/cnlohr/esp8266ws2812i2s
Sorry for the confusion.

Hardware/code seems to mostly work, but the level shifter seems to be required to make it work. Scope output looks ok but initially the lights are not 100% stable, they seem to calm down and become stable so something it still not completely correct. Either the WS2811/WS2812 have differing tolerances on the timing or more likely I have not got the hardware quite optimal.