cnlohr/ws2812esp8266 #17: Insecure

jonshouse1, 2016-06-05:
Connecting directly to the ESP module allows a user to extract the password for an access point.

The password should be blank, the user should have to re-enter it before "change settings"

jonshouse1, 2016-06-05:
My mistake, seems once in station mode it no longer has an active AP ? 

It might be worth changing the html to stop the password being displayed.

(Disclaimer not tested)
<tr><td>PASS:</td><td><input type=\"text\" id=\"wificurpassword\"></td></tr> \
To:
<tr><td>PASS:</td><td><input type=\"password\" id=\"wificurpassword\"></td></tr> \

cnlohr, 2016-06-06:
This is a really good idea.  You could change this... If you don't mind, also make a checkbox to 'unmask'  I don't know how to do that, but would be cool to give user access if they wanted?