cnlohr/ws2812esp8266 #18: Not an issue, just a feature request :-)

jonshouse1, 2016-06-05:
It would be great if the esp code could have an offset for the first pixel as a field in the web front end.

For example I have 4 modules with 50 LEDs per module, I can tell the first module its first pixel is at offset 0, it now 'owns' pixels 0 through 49.
The second module has its first pixel at offset 50 and so on.  The UDP broadcast packet has the values for all 4 sets of 50. 

Also a chance to make the first pixel seem to be offset 0 rather than 3.
This would allow the esp driver to work with some existing packet formats as they tend to have chunks of 8 bit values after a header, the offset would allow the user to ignore the header bytes.

Thanks,
Jon

jonshouse1, 2016-06-05:
PS I forget to mention user also need to able to change the listening port to make this sensible...

jonshouse1, 2016-06-09:
I need to understand the code but I will give it a go.

jonshouse1, 2016-06-10:
https://github.com/jonshouse1/esp8266ws2812i2s_JA/tree/master/user

Ok, made a start.  I find your code is tad complex on the http side. Not sure what the union structure for user data is about, all seems a little tricky to me.I normally define a single struct 'config', mark it packed and 32 bit boundry aligned then load and save that to flash.  I am still trying to figure out how your http userdata code works before I mod the front end.  All untested at the moment !

cnlohr, 2016-06-12:
Sorry, sent back to you in email, but should probably say here.

UserData in the httpstruct has nothing to do with UserData in the SETTING struct.  That UserData is wide open for your use!!! Feel free to create a MyUserData Structure or something and just typecast the SETTINGS.UserData structure to your whim!