cnlohr/ws2812esp8266 #21: Not an issue, but clarification needed

ArchontPL, 2016-10-05:
Hi,
I'd like to ask about SEND_WS_0() function. I know how ws2812 work and I get idea of i.e. this line
`time = 3; while(time--) WRITE_PERI_REG( PERIPHS_GPIO_BASEADDR + GPIO_ID_PIN(WSGPIO), 1 );`
but I don't see any delay in this while loop. So this must mean that 3 execution of WRITE_PERI_REG function take about 0.35us (T0H from ws2812 datasheet) and 8 execution of WRITE_PERI_REG take 0.8us (T0L from ws2812 datasheet) , right?

So approximately one execution of WRITE_PERI_REG take 8 CPU cycles ( 0.1us == 10MHz, ESP2866 is running with 80MHz) ?

cnlohr, 2016-10-05:
That takes waaay more than 8 cpu cycles!  There's 1 write buffer, so it's probably more like 60 something cpu cycles!  Only instructions that deal inside the CPU core are about 1 clock per cycle.  RAM is typically 2, FLASH 5 or 6, other IO even more, yet!

I do ets_intr_lock() to disable interrupts that could mess up timing.

I don't know what you're referring to with the 3rd thing.

ArchontPL, 2016-10-25:
@cnlohr at third thing I mean some kind of datasheet which describe avilable peripherals.

cnlohr, 2016-11-06:
I don't really know what to offer you.  The only way I find anything is by googling, a lot.  I don't have any links to datasheets or peripheral lists or anything like that.  I just google keywords like esp8266 i2s or stuff like that.