cnlohr/ws2812esp8266 #24: Build error on final step - libcrypto.a: No such file or directory

sympthom9, 2017-05-28:
Hi,

Thanks for some great content both on your channel on youtube and here on github!

I am getting a build error as you can see:

sympthom@ubuntu:~/src/esp8266/esp8266ws2812i2s$ make burn
esp82xx/common.mf:39: Warning: No shell variable 'ESP_ROOT', using '/home/sympthom/esp8266/esp-open-sdk'
Makefile:22: warning: overriding recipe for target 'purge'
esp82xx/main.mf:158: warning: ignoring old recipe for target 'purge'
~/src/esp8266/crosstool-NG/builds/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -Os -I~/src/esp8266/esp_iot_sdk_v0.9.3/include -Iesp82xx/include -I. -Iesp82xx/fwsrc -Iuser -DICACHE_FLASH -DSK6812 -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50 -DVERSSTR='"Version: v0.5-27-gd26c4-dev - Build Mon, May 29 2017, 00:47:51 +0200 with -DICACHE_FLASH -DSK6812 -DWEB_PORT=80 -DCOM_PORT=7777 -DBACKEND_PORT=7878  -DSLOWTICK_MS=50"' esp82xx/fwsrc/uart.c esp82xx/fwsrc/esp82xxutil.c esp82xx/fwsrc/flash_rewriter.c esp82xx/fwsrc/http.c esp82xx/fwsrc/commonservices.c esp82xx/fwsrc/http_custom.c esp82xx/fwsrc/mdns.c esp82xx/fwsrc/mfs.c user/custom_commands.c user/user_main.c user/ws2812_i2s.c user/pattern.c -flto -Wl,--relax -Wl,--gc-sections -nostdlib -L~/src/esp8266/esp_iot_sdk_v0.9.3/lib -L~/src/esp8266/crosstool-NG/builds/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/libmain.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/liblwip.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/libssl.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/libupgrade.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/libnet80211.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/libwpa.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/libphy.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/liblwip.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/libcrypto.a ~/src/esp8266/esp_iot_sdk_v0.9.3/lib/libpp.a ~/src/esp8266/crosstool-NG/builds/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/libgcc.a -T esp82xx/toolchain/ld/linkerscript.ld -T ~/src/esp8266/esp_iot_sdk_v0.9.3/ld/eagle.rom.addr.v6.ld -B~/src/esp8266/esp_iot_sdk_v0.9.3/lib  -o image.elf
xtensa-lx106-elf-gcc: error: /home/sympthom/src/esp8266/esp_iot_sdk_v0.9.3/lib/libcrypto.a: No such file or directory
esp82xx/main.mf:85: recipe for target 'image.elf' failed
make: *** [image.elf] Error 1

Any clues what to do?

Thanks and keep up the great work!

//Martin

cnlohr, 2017-05-29:
This project is kept only for archival purposes and support has terminated.  Sorry :-/. Please use esp8266ws2812i2s.  

One of the big things though is you can't really use the 0.9.3 SDK anymore, it has been long since obsoleted :-/

closed 2017-05-29 by cnlohr

sympthom9, 2017-05-29:
Don't be sorry when it's great news a new repository superseeds and there is active development. Also I need to learn how to read the readme where you mentioned at the very beginning that the repository was obsolete and discontionued.

I cloned the repository you mentioned, work great!

Thanks hacker!